DJ2LS/FreeDATA #837: GUI adjustments needed

DJ2LS, 2024-10-21:
[image]

DJ2LS, 2024-10-21:
[image]

[image]

dk5sm, 2024-10-21:
[image]

Could see some states with missing callsign here also.

DJ2LS, 2024-11-09:
> ![image](https://private-user-images.githubusercontent.com/58949681/378413012-0c7cb9ed-9b1a-4557-a51a-f99bb8b9776b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzExNDg1NTgsIm5iZiI6MTczMTE0ODI1OCwicGF0aCI6Ii81ODk0OTY4MS8zNzg0MTMwMTItMGM3Y2I5ZWQtOWIxYS00NTU3LWE1MWEtZjk5YmI4Yjk3NzZiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA5VDEwMzA1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFmNTc2ZWM2YmQxNDFkNDE1MjE3MzYzMTVlY2ExMzE1ZDYzNGQxMjdkNjFhNTAzMWNhZDdmYmYwZDdkZWJiODMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.APsZNiiWX-lEm-pwGF1lYKCCzLocg1aXz1wBrc7XEbI)
> 
> Could see some states with missing callsign here also. Guess this is related to ARQ_BURST_ACK, ARQ_SESSION_INFO_ACK and ARQ_BURST_FRAME only.

done

closed 2024-11-10 by DJ2LS