DJ2LS/FreeDATA #846: [Bug]: GUI: status bar not shown on Android

dk5sm, 2024-11-07:
### FreeDATA Version

0.16.9-alpha

### What operating system are you using?

Other (specify below)

### Operating System Version

Android 13 / MS Edge (chrome)

### Architecture

x86

### Area of problem

GUI

### Describe the bug

GUI on Android Tablet or Smartphone is always missing the status bar. Horizontal or vertical alignment does not seem to play a role here:

[image]

### To Reproduce

Run FD GUI on Android device

### Expected behavior

It should be displayed as intended (as it does on Linux/Windows and on iOS).

DJ2LS, 2025-03-19:
[image]

Thats a test result of today. Might be device specific?

dk5sm, 2025-03-19:
It seems that you are using Android´s stock browser. This one works for me also, but using Chrome, Edge or FF still shows no footer.

DJ2LS, 2025-03-19:
Because of poor performance of my Android VM it would be a pain installing them.

DJ2LS, 2025-03-20:
Thanks to Lucas, my friendly neighbour, I have a tablet for reproducing this issue. 😃

DJ2LS, 2025-03-20:
Enjoy your tablet...please give me feedback

closed 2025-04-05 by DJ2LS