DJ2LS/FreeDATA #854: [Bug]: GUI error in modify config

frspin, 2024-12-03:
### FreeDATA Version

0.16.10.alpha

### What operating system are you using?

Ubuntu

### Operating System Version

Ubuntu 24.04 full up to date

### Architecture

x86

### Area of problem

GUI

### Describe the bug

In .ini file respond_to_cq is set to True
In GUI it is shown as False.

2024-12-03 11:30:28 [info     ] [API] POST                     response_code=200 url=http://127.0.0.1:5000/config
2024-12-03 11:30:51 [info     ] [EXPLORER] Data pushed successfully
{'NETWORK': {'modemaddress': '127.0.0.1', 'modemport': 5000}, 'STATION': {'mycall': 'IW2DHW', 'mygrid': 'JN45op', 'ssid_list': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9], 'myssid': 0, 'enable_explorer': True, 'enable_stats': True, 'respond_to_cq': True, 'enable_callsign_blacklist': False, 'callsign_blacklist': []}, 'AUDIO': {'input_device': '5b4d', 'output_device': '5b4d', 'rx_audio_level': 0, 'tx_audio_level': 0}, 'RIGCTLD': {'ip': '127.0.0.1', 'port': 4532, 'path': '', 'command': '', 'arguments': '', 'enable_vfo': False}, 'RADIO': {'control': 'rigctld', 'model_id': 1043, 'serial_port': '/dev/ttyUSB0', 'serial_speed': 38400, 'data_bits': 8, 'stop_bits': 1, 'serial_handshake': 'ignore', 'ptt_port': 'ignore', 'ptt_type': 'RIG', 'serial_dcd': 'NONE', 'serial_dtr': 'OFF', 'serial_rts': 'OFF'}, 'TCI': {'tci_ip': '127.0.0.1', 'tci_port': 50001}, 'MESH': {'enable_protocol': False}, 'MODEM': {'enable_hmac': False, 'enable_morse_identifier': False, 'tx_delay': 50, 'maximum_bandwidth': 2438, 'enable_socket_interface': False, 'respond_to_cq': True}, 'SOCKET_INTERFACE': {'enable': False, 'host': '127.0.0.1', 'cmd_port': 8000, 'data_port': 8001}, 'MESSAGES': {'enable_auto_repeat': False, 'adif_log_host': '', 'adif_log_port': 0}, 'GUI': {'auto_run_browser': True}}
{'NETWORK': {'modemaddress': '127.0.0.1', 'modemport': 5000}, 'STATION': {'mycall': 'IW2DHW', 'mygrid': 'JN45op', 'ssid_list': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9], 'myssid': 0, 'enable_explorer': True, 'enable_stats': True, 'respond_to_cq': True, 'enable_callsign_blacklist': False, 'callsign_blacklist': []}, 'AUDIO': {'input_device': '5b4d', 'output_device': '5b4d', 'rx_audio_level': 0, 'tx_audio_level': 0}, 'RIGCTLD': {'ip': '127.0.0.1', 'port': 4532, 'path': '', 'command': '', 'arguments': '', 'enable_vfo': False}, 'RADIO': {'control': 'rigctld', 'model_id': 1043, 'serial_port': '/dev/ttyUSB0', 'serial_speed': 38400, 'data_bits': 8, 'stop_bits': 1, 'serial_handshake': 'ignore', 'ptt_port': 'ignore', 'ptt_type': 'RIG', 'serial_dcd': 'NONE', 'serial_dtr': 'OFF', 'serial_rts': 'OFF'}, 'TCI': {'tci_ip': '127.0.0.1', 'tci_port': 50001}, 'MESH': {'enable_protocol': False}, 'MODEM': {'enable_hmac': False, 'enable_morse_identifier': False, 'tx_delay': 50, 'maximum_bandwidth': 2438, 'enable_socket_interface': False, 'respond_to_cq': True}, 'SOCKET_INTERFACE': {'enable': False, 'host': '127.0.0.1', 'cmd_port': 8000, 'data_port': 8001}, 'MESSAGES': {'enable_auto_repeat': False, 'adif_log_host': '', 'adif_log_port': 0}, 'GUI': {'auto_run_browser': True}}

If I change it to True with GUI FD show an error:

Exception in ASGI application
  + Exception Group Traceback (most recent call last):
  |   File "/usr/lib/python3/dist-packages/starlette/_utils.py", line 82, in collapse_excgroups
  |     yield
  |   File "/usr/lib/python3/dist-packages/starlette/middleware/base.py", line 190, in __call__
  |     async with anyio.create_task_group() as task_group:
  |   File "/usr/lib/python3/dist-packages/anyio/_backends/_asyncio.py", line 678, in __aexit__
  |     raise BaseExceptionGroup(
  | ExceptionGroup: unhandled errors in a TaskGroup (1 sub-exception)
  +-+---------------- 1 ----------------
    | Traceback (most recent call last):
    |   File "/usr/lib/python3/dist-packages/uvicorn/protocols/http/h11_impl.py", line 412, in run_asgi
    |     result = await app(  # type: ignore[func-returns-value]
    |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |   File "/usr/lib/python3/dist-packages/uvicorn/middleware/proxy_headers.py", line 84, in __call__
    |     return await self.app(scope, receive, send)
    |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |   File "/usr/lib/python3/dist-packages/fastapi/applications.py", line 289, in __call__
    |     await super().__call__(scope, receive, send)
    |   File "/usr/lib/python3/dist-packages/starlette/applications.py", line 116, in __call__
    |     await self.middleware_stack(scope, receive, send)
    |   File "/usr/lib/python3/dist-packages/starlette/middleware/errors.py", line 186, in __call__
    |     raise exc
    |   File "/usr/lib/python3/dist-packages/starlette/middleware/errors.py", line 164, in __call__
    |     await self.app(scope, receive, _send)
    |   File "/usr/lib/python3/dist-packages/starlette/middleware/cors.py", line 91, in __call__
    |     await self.simple_response(scope, receive, send, request_headers=headers)
    |   File "/usr/lib/python3/dist-packages/starlette/middleware/cors.py", line 146, in simple_response
    |     await self.app(scope, receive, send)
    |   File "/usr/lib/python3/dist-packages/starlette/middleware/base.py", line 189, in __call__
    |     with collapse_excgroups():
    |   File "/usr/lib/python3.12/contextlib.py", line 158, in __exit__
    |     self.gen.throw(value)
    |   File "/usr/lib/python3/dist-packages/starlette/_utils.py", line 88, in collapse_excgroups
    |     raise exc
    |   File "/usr/lib/python3/dist-packages/starlette/middleware/base.py", line 191, in __call__
    |     response = await self.dispatch_func(request, call_next)
    |                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 109 lines cut ...]
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/fastapi/routing.py", line 190, in run_endpoint_function
    return await dependant.call(**values)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/Prog/ham/FreeDATA/freedata_server/server.py", line 377, in post_config
    set_config = app.config_manager.write(config)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/Prog/ham/FreeDATA/freedata_server/config.py", line 201, in write
    self.validate_data(data)
  File "/mnt/Prog/ham/FreeDATA/freedata_server/config.py", line 129, in validate_data
    if not isinstance(data[section][setting], self.config_types[section][setting]):
                                              ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
KeyError: 'respond_to_cq'

After this modem continue to run

### To Reproduce

Start FD
Go to Config tab
Try to change respond_to_cq parameter

### Expected behavior

A correct config show and, if necessary, change

DJ2LS, 2024-12-05:
Hi @frspin , thanks for the report. 

There is an configuration related inconsistency.

It should look like this:

```
[NETWORK]
modemaddress = 127.0.0.1
modemport = 5000

[STATION]
mycall = AA1AAA
mygrid = JN48ea
myssid = 1
ssid_list = [0, 1, 2, 3, 4, 5, 6, 7, 8, 9]
enable_explorer = False
enable_stats = False
respond_to_cq = True
enable_callsign_blacklist = False
callsign_blacklist = []

[AUDIO]
input_device = 5a1c
output_device = bd6c
rx_audio_level = 0
tx_audio_level = 0

[RIGCTLD]
ip = 127.0.0.1
port = 4532
path = 
command = 
arguments = 
enable_vfo = False

[RADIO]
control = disabled
model_id = 1001
serial_port = /dev/cu.Bluetooth-Incoming-Port
serial_speed = 38400
data_bits = 8
stop_bits = 1
serial_handshake = ignore
ptt_port = ignore
ptt_type = USB
serial_dcd = NONE
serial_dtr = OFF
serial_rts = OFF

[TCI]
tci_ip = 127.0.0.1
tci_port = 50001

[MESH]
enable_protocol = False

[MODEM]
enable_hmac = False
enable_morse_identifier = False
tx_delay = 50
maximum_bandwidth = 2438
enable_socket_interface = False

[SOCKET_INTERFACE]
enable = False
host = 127.0.0.1
cmd_port = 8000
data_port = 8001

[MESSAGES]
enable_auto_repeat = False
adif_log_host = 127.0.0.1
adif_log_port = 2237

[GUI]
auto_run_browser = True

```

In other words: remove `respond_to_cq` from `MODEM` part of the config. This should solve the problem with the key error.

frspin, 2024-12-05:
Il 05/12/24 10:27, DJ2LS ha scritto:
> Hi @frspin <https://github.com/frspin> , thanks for the report.
>
> There is an configuration related inconsistency.
>
> It should look like this:
>
> |[NETWORK] modemaddress = 127.0.0.1 modemport = 5000 [...]

In my config file respond_to_cq is only in [STATION] section and not in [MODEM] section.

[NETWORK]
modemaddress = 127.0.0.1
modemport = 5000

[STATION]
mycall = IW2DHW
mygrid = JN45op
ssid_list = [0, 1, 2, 3, 4, 5, 6, 7, 8, 9]
myssid = 0
enable_explorer = True
enable_stats = True
respond_to_cq = True
enable_callsign_blacklist = False
callsign_blacklist = []

[AUDIO]
input_device = 5b4d
output_device = 5b4d
rx_audio_level = 0
tx_audio_level = 0

[RIGCTLD]
ip = 127.0.0.1
port = 4532
path =
command =
arguments =
enable_vfo = False

[RADIO]
control = rigctld
model_id = 1043
serial_port = /dev/ttyUSB0
serial_speed = 38400
data_bits = 8
stop_bits = 1
serial_handshake = ignore
ptt_port = ignore
ptt_type = RIG
serial_dcd = NONE
serial_dtr = OFF
serial_rts = OFF

[TCI]
tci_ip = 127.0.0.1
tci_port = 50001

[MESH]
enable_protocol = False

[MODEM]
enable_hmac = False
enable_morse_identifier = False
tx_delay = 50
maximum_bandwidth = 2438
enable_socket_interface = False

[SOCKET_INTERFACE]
enable = False
host = 127.0.0.1
cmd_port = 8000
data_port = 8001

[MESSAGES]
enable_auto_repeat = False
adif_log_host =
adif_log_port = 0

[GUI]
auto_run_browser = True

Franco Spinelli
IW2DHW

DJ2LS, 2024-12-05:
Thanks, @frspin 

okay, another option is, your gui is pushing data in the wrong order. So, did you run "`npm run build`" with the latest code?

frspin, 2024-12-05:
Il 05/12/24 11:14, DJ2LS ha scritto:
> Thanks, @frspin <https://github.com/frspin>
> 
> okay, another option is, your gui is pushing data in the wrong order. So, did you run "|npm run 
> build|" with the latest code?
> 

Why in server.py, at line 240, there is a "respond_to_cq" key (unused?) in MODEM section?
And why "respond_to_cq" key is shown in "modem" tab for config instead of "station" tab?

Franco Spinelli
IW2DHW

DJ2LS, 2024-12-05:
> Il 05/12/24 11:14, DJ2LS ha scritto:
> Thanks, @frspin <https://github.com/frspin> okay, another option is, your gui is pushing data in the wrong order. So, did you run "|npm run build|" with the [...]

this one is a typo, it should be part of the STATION settings. I changed it, thanks. However, this shouldn't affect settings handling.

DJ2LS, 2024-12-05:
Maybe your code is correctly, but the browser is holding an outdated gui version in cache? Can you delete it or using an alternative browser for validation?

DJ2LS, 2024-12-05:
@frspin latest development code now contains a attempt with disabled caching. Maybe this helps as well.

frspin, 2024-12-05:
Il 05/12/24 13:41, DJ2LS ha scritto:
> Maybe your code is correctly, but the browser is holding an outdated gui version in cache? Can you 
> delete it or using an alternative browser for validation?

Opened GUI in Chrome and aspect (as seen in attached image) is different.

And now this switch is correctly shown in "Station" tab.

Cleaned also the cache of Firefox and now also in Firefox the aspect is correct.

Also updated to latest development code and now there is no cache from 127.0.0.1

Franco Spinelli
IW2DHW

DJ2LS, 2024-12-09:
So it seems, this issue can be closed?

frspin, 2024-12-09:
Il 09/12/24 09:42, DJ2LS ha scritto:
> So it seems, this issue can be closed?
> 

Yes

Franco Spinelli
IW2DHW

closed 2024-12-09 by DJ2LS