DJ2LS/FreeDATA #856: [Bug]: FreeDATA seems to send a "broken' ADIF when using Wavelog Gate

erikdewildt, 2024-12-15:
### FreeDATA Version

0.16.10-alpha

### What operating system are you using?

Ubuntu

### Operating System Version

Ubuntu 24.10

### Architecture

x86

### Area of problem

Server - General

### Describe the bug

When using the `ADIF` button in the chat window I get an error in Wavelog-Gate: `Received broken ADIF`.

I'm using Wavelog-Gate version 1.0.20 to send my logging towards my Wavelog logbook. I can't find any more specific error in my logs unfortunately.

### To Reproduce

1. Go to a chat window
2. Click on the ADIF button
3. Observe the Wavelog Gate window to display an error.

### Expected behavior

It would be great if I could send logs from FreeDATA to Wavelog as both can be used remotely using a web browser.

### Screenshots

[image]

DJ2LS, 2024-12-15:
Thanks for the report, @erikdewildt . Could you do me the favour, sending a adif string which is expected by wave log? There seems to be a difference of expected string for different log software.

DJ2LS, 2024-12-15:
Nevermind, I think it should be valid adif, regardless of the software. 
this bug might need some time to solve.

erikdewildt, 2024-12-16:
I first thought that the issue is that Wavelog-Gate does not expect a "ADIF" record. The docs show that I should use the "Secondary UDP Server". I was thinking that this might use an other format. So I tried to capture the logging from WSJT-X.

When I send a message from WSJT-X using the main log server option the output is this:

`<adif_ver:5>3.1.0
<programid:6>WSJT-X
<EOH>
<call:6>PE1PUX <gridsquare:0> <mode:3>FT8 <rst_sent:0> <rst_rcvd:0> <qso_date:8>20241216 <time_on:6>122324 <qso_date_off:8>20241216 <time_off:6>122324 <band:3>40m <freq:8>7.075500 <station_callsign:6>PE1PUX <my_gridsquare:6>JO21WA <EOR>
`

When  I use the secondary udp server the output is:

`<call:6>PE1PUX <gridsquare:0> <mode:3>FT8 <rst_sent:0> <rst_rcvd:0> <qso_date:8>20241216 <time_on:6>122732 <qso_date_off:8>20241216 <time_off:6>122732 <band:3>40m <freq:8>7.054500 <station_callsign:6>PE1PUX <my_gridsquare:6>JO21WA <eor>`

Using the same script when capturing the ADIF broadcast from FreeDATA the output looks like:

`<CALL:5>DJ2LS<QSO_DATE:8>20241215<TIME_ON:6>140424<TIME_OFF:6>140446<FREQ:5>7.053<MODE:7>DYNAMIC<SUBMODE:8>FREEDATA<COMMENT:24>QSO via FreeDATA RF-Chat<GRIDSQUARE:4>----<EOR>`

It seems that FreeDATA looks quite similar to the WSJT-X output using the Secondary UDP option of WSJT-X, the format WavelogGate expects. Besides the capital letters in tags only thing I noticed was the `----` as gridsquare.
Using the `ADIF` button on an other QSO gave this output:

`<CALL:6>OK1MDX<QSO_DATE:8>20241215<TIME_ON:6>150020<TIME_OFF:6>150634<FREQ:5>7.053<MODE:7>DYNAMIC<SUBMODE:8>FREEDATA<COMMENT:24>QSO via FreeDATA RF-Chat<GRIDSQUARE:6>JO70FA<EOR>`

And yes, that is something which WavelogGate accepts and logs to WaveLog:

[image]

[image]

So as long as the QSO data is valid logging seems to work. Issue was that on my attempt I did not seem to have received your QTH locator.

I leave it up to you if you would like to keep this ticket open to improve things in future releases of close it for now.

Thanks for all the work you put in to this. I've started looking into the code as I noticed u are using FastAPI and I have some experience using FastAPI myself :-)

73 Erik

DJ2LS, 2024-12-16:
Thanks for your detailed feedback, @erikdewildt ! To me it seems, the issue might be the gridsquare. If `----` is used, logging fails? The `----` is a placeholder in FreeDATA, means no gridsquare data is available. So the question is, what would be a valid placeholder. Or should we remove it if no valid data is available?

DJ2LS, 2024-12-17:
A check on wave log gates source code shows, that there is an ADIF parser which checks if it is valid or not. So I assume we have to remove the placeholder grid square.

DJ2LS, 2025-01-21:
@LA3QMA this is the corresponding issue regarding the adif bug i mentioned

LA3QMA, 2025-01-24:
I have tried several things and could not get WaveLogGate to accept anything. The format from wsjt is also deprecated.
Can you try https://github.com/DJ2LS/FreeDATA/tree/develop_kgb 

If so you can add QSOs to Wavelog without WaveLogGate. In the Chat config you can add the URL:port and the Wavelog API_key

erikdewildt, 2025-01-24:
Hi,

I must be doing something wrong. I did a checkout of that branch but I don't have any option in the Chat config to add the wavelog URL and API key.

`
erik@PE1PUX:/opt/freedata/FreeDATA$ git status
On branch develop_kgb
Your branch is up to date with 'origin/develop_kgb'.

nothing to commit, working tree clean
`

[image]

If I check the code on my machine I can see the changes you made in de commit for adding the Wavelog configuration parameters. I use the `run-freedata-linux.sh` script to start the application. But, the GUI seems not to reflect your changes.

Any clues on what i am doing wrong? I would love to test this feature :-)

Thanks,
Erik

LA3QMA, 2025-01-24:
check if your freedata_gui/src/components/settings_chat.vue is reference to wavelog

If so: go into the folder: FreeDATA/freedata_gui and enter:
npm i && npm run build

Then re-start FreeDATA.

erikdewildt, 2025-01-24:
It seems to work. I disabled (removed) the `ADIF Log Host` and provided the WaveLog details. When pressing the `ADIF` button on a chat a new entry appears in my WaveLog Instance.

Thanks for your work on this!

LA3QMA, 2025-01-24:
The config.ini should have two new settings under [MESSAGES]
wavelog_host = http://localhost:8086
wavelog_api_key = something

I'll keep the issue open a little longer in case you find some other issues with my try to implement Wavelog.

DJ2LS, 2025-01-29:
should be fixed in develop branch and part of a new release.

closed 2025-01-29 by DJ2LS