DJ2LS/FreeDATA #862: [Feature Request]: Open URL Links in new tab

mprilepok, 2025-01-19:
### Problem Description

HI.
When you click on the URL link from chat it opens in the current tab. Please provide a configuration option to set if the URL link in chat should be opened in the current tab or a new one.

### Proposed Solution

Provided a configuration how should the URL be opened- current tab or a new one.

### Alternatives Considered

Current state - not prefered.

DJ2LS, 2025-01-21:
Thanks for the feedback. 

In my opinion it's making sense if a url is always openend in a new tab.

DJ2LS, 2025-01-21:
@LA3QMA are you interested in trying to fix this? It's mainly JavaScript / HTML, maybe just a small parameter

closed 2025-01-31 by DJ2LS