DJ2LS/FreeDATA #882: [Feature Request]: 500 Hz bandwidth limit when AFK

ND2W, 2025-02-13:
### Problem Description

According to FCC regulation 97.221 (c) (2), transmissions from an automatically controlled station cannot occupy a bandwidth of more than 500 Hz.
www.ecfr.gov/current/title-47/section-97.221

Currently FD does not change the bandwidth limit automatically when the user indicates that they are away from keyboard (which would mean the control operator has left the station and the station is now operating automatically). 

A user would indicate that they are away from keyboard by activating the "Away From Key" switch in a widget.

### Proposed Solution

1. Change the FreeData bandwidth limit to 500 Hz (under Settings > Modem > Maximum used bandwidth) when the user activates the "Away From Key" switch.
2. (a) Maybe update the widget label "Away From Key" to "Away From Key - 500 Hz bandwidth limit" or similar to inform the user that the bandwidth limit, when "Away From Key" is activated, will be reduced to 500 Hz. 
2. (b) Maybe add this information to under Settings > Modem > Maximum used bandwidth to the pop-up that shows when the mouse hovers over the "?". The following text could be added: "The bandwidth limit will automatically be set to 500 Hz if the user activates the "Away From Key" switch to comply with FCC regulation § 97.221 Automatically controlled digital station."

### Alternatives Considered

N/A

DJ2LS, 2025-02-15:
Thanks for the issue! I think a first start would be a "help button" like we have them in settings, for informing the user about the law related situation and what's behind "afk" . 

reducing bandwidth down to 500Hz could be done as a setting on server side. "Set bandwidth 500Hz when AFK".

petrkr, 2025-02-15:
Is that related only to USA or it is world wide?
I just wonder to not make it just because of one region from many. If so, then I suggest some checkbox "Use 500Hz max while AFK" in settings instead force everyone on world to be on 500Hz just because of few countries have such regulation.

DJ2LS, 2025-02-15:
@petrkr yes, that's what I meant with making it a setting, so the corresponding user of a specific area can decide by their own

DJ2LS, 2025-03-17:
Is this a critical feature for 1.0?

ND2W, 2025-03-17:
Not critical in my opinion, as the bandwidth can be set manually.