DJ2LS/FreeDATA #883: [Feature Request]: Make frequency widtech font adjustable

petrkr, 2025-02-15:
### Problem Description

That is not problem, just feature request.

Make Frequency widget able to change font size

[image]

### Proposed Solution

Add some font selector to frequency widget

### Alternatives Considered

Zoom in chrome, but that will zoom everything else too

DJ2LS, 2025-02-15:
Thanks for the feature request. We have this also for some other widgets. I have been trying to work on this, but wasn't able fixing it in an easy way. Maybe I have been on the wrong path. I'm wondering, if bootstrap framework has something built in.

DJ2LS, 2025-03-17:
This seems to be more tricky causing more code overhead than expected. Maybe I'm in the wrong way. Help is appreciated with a working code snippet maybe.