DJ2LS/FreeDATA #895: [Feature Request]: Save server log file

OH2JRB, 2025-02-25:
### Problem Description

On Windows platform, by default, the server log file is not saved into file.

### Proposed Solution

Modify Windows application launcher file in the Windows distribution kit. Otherwise, updating application is likely to override any local modifications.

### Alternatives Considered

Local modification. But may be needed every time user installs new version from GitHub.

### Additional Information

Info in server log file could be useful in troubleshooting.

DJ2LS, 2025-02-26:
Thanks, @OH2JRB , problem with logging is, hard errors are not included in logging. But they are the real interesting information. So I have no idea right now how we could save the, as well somehow.

DJ2LS, 2025-10-30:
Should be working now, I'm closing this one, please reopen it, if its not working

closed 2025-10-30 by DJ2LS