DJ2LS/FreeDATA #914: [Bug]: Darkmode: Buttons for load and save preset not visable

dominik0815, 2025-03-16:
### FreeDATA Version

0.17.0

### What operating system are you using?

Windows

### Operating System Version

Windows 10

### Architecture

x64

### Area of problem

GUI

### Describe the bug

While usage of darkmode the "load preset" and "save preset" buttons at the "add sections"-menu are not visible. Something is highlighted but if cursor is not on top there is no information about the buttons.

### To Reproduce

-activate dark mode
-try load or save preset

### Expected behavior

Restore the visibility of the buttons.

### Screenshots

[image]

dominik0815, 2025-03-16:
Also the contrast of the gui in dark mode is not high enough. In special the horizontal lines (black) at the data rate / SNR section is hardly to see. Maybe the background colour of these sections could be changed to more bright grey.

[image]

DJ2LS, 2025-03-17:
Fixed