DJ2LS/FreeDATA #922: [Bug]: Wiki macOS Install Path wrong

HB9HBO, 2025-03-19:
### FreeDATA Version

any

### What operating system are you using?

Other (specify below)

### Operating System Version

macOS Install in Wiki

### Architecture

x86

### Area of problem

Installation

### Describe the bug

The download path for run-freedata-macos.sh and install-freedata-macos.sh leads to a wrong location, so curl gets a 404 error.

Maybe it should be "develop/tools/macOS" instead of "main"

### To Reproduce

curl -o run-freedata-macos.sh https://raw.githubusercontent.com/DJ2LS/FreeDATA/main/run-freedata-macos.sh

### Expected behavior

curl -o run-freedata-macos.sh https://raw.githubusercontent.com/DJ2LS/FreeDATA/develop/tools/macOS/run-freedata-macos.sh

curl -o install-freedata-macos.sh https://raw.githubusercontent.com/DJ2LS/FreeDATA/develop/tools/macOS/install-freedata-macos.sh

DJ2LS, 2025-03-19:
Thanks, @HB9HBO !

@LA3QMA , could you change this in wiki? I suggest we are going to keep the main branch as we are going to merge the next days

DJ2LS, 2025-03-19:
@HB9HBO Would it be okay for you, if we are going to close this, if it's fixed? ( Even if we have the scripts not yet merged into main? ) This will happen the next days as soon as we have finished #835

HB9HBO, 2025-03-19:
For shure.
Thank you :)

closed 2025-03-19 by DJ2LS