DJ2LS/FreeDATA #926: [Bug]: Blacklist not working as expected

LA3QMA, 2025-03-27:
### FreeDATA Version

develop

### What operating system are you using?

Windows

### Operating System Version

win10

### Architecture

x64

### Area of problem

Server - Message System

### Describe the bug

Testing with LA3QMA-1 and added "N0CALL" to the blacklist.

When N0CALL-1 transmit a beacon the receiving station doesn't show anything in "Heard stations", "Activity" or pop-up. This is as expected
When N0CALL-1 transmit a CQ, LA3QMA-1 doesn't respond to this as expected.
When N0CALL-1 transmit a message to LA3QMA-1, LA3QMA-1 doesn't respond to the request as expected.

When LA3QMA-1 transmit a CQ, N0CALL-1 responds and LA3QMA-1 shows a pop-up with "CQ received: LA3QMA-1" and "QRV received: LA3QMA-1" - this should be dropped and it also shows the wrong callsign.

When LA3QMA-1 pings N0CALL-1, N0CALL-1 responds as expected, LA3QMA-1 receive the respond but doesn't show any pop-up and nothing in Heard/Activity - as expected

When LA3QMA-1 start a new message with N0CALL-1 the message fails as expected.

### To Reproduce

Testing with LA3QMA-1 and added "N0CALL" to the blacklist.

When N0CALL-1 transmit a beacon the receiving station doesn't show anything in "Heard stations", "Activity" or pop-up. This is as expected
When N0CALL-1 transmit a CQ, LA3QMA-1 doesn't respond to this as expected.
When N0CALL-1 transmit a message to LA3QMA-1, LA3QMA-1 doesn't respond to the request as expected.

When LA3QMA-1 transmit a CQ, N0CALL-1 responds and LA3QMA-1 shows a pop-up with "CQ received: LA3QMA-1" and "QRV received: LA3QMA-1" - this should be dropped and it also shows the wrong callsign.

When LA3QMA-1 pings N0CALL-1, N0CALL-1 responds as expected, LA3QMA-1 receive the respond but doesn't show any pop-up and nothing in Heard/Activity - as expected

When LA3QMA-1 start a new message with N0CALL-1 the message fails as expected.

### Expected behavior

The station with an active blacklist should not show anything to/from stations in the blacklist.

DJ2LS, 2025-03-28:
@LA3QMA I pushed a fix, if you find the time, could you test again?

LA3QMA, 2025-03-28:
Seems to work as expected now