DJ2LS/FreeDATA #941: high CPU usage with beacons activated

ko0ooo, 2025-04-12:
CPU using either Chrome or Firefox increases approximately 15% when beacons are activated.

Computer is a Pi4, using Chrome w/beacons activated

[image]

and beacons deactivated

[image]

 Richard    ko0ooo

ko0ooo, 2025-04-12:
forgot to mention.. .  using version 17.1 on a Pi4

Richard   ko0ooo

Mashintime, 2025-04-12:
Tried clearing/restoring a grid default?  Are you using the animated beacon button.

Mashintime, 2025-04-12:
If you don't used the animated button, does the extra cpu usage go away?

ko0ooo, 2025-04-12:
it sure does...    deleted the animated beacon widget and used the beacon function with the 'Broadcasts' widget.

Richard   ko0ooo

[image]

Mashintime, 2025-04-12:
On PC i don't see any additional CPU usage with the animated widget; so assuming something specific to waylan on the Pi.  I'd contine to use the non-animated widget.

DJ2LS, 2025-04-16:
So the problem exists only with the small beacon widget, right?

DJ2LS, 2025-04-23:
Is this Issue still active or can we close it?