DJ2LS/FreeDATA #960: [Bug]: Unable To Use FL-RIG

KhunDoug, 2025-04-20:
### FreeDATA Version

0.17 Beta

### What operating system are you using?

Windows

### Operating System Version

Windows 11 24H2

### Architecture

x64

### Area of problem

Server - Message System

### Describe the bug

Attempting to get rig control using FL-RIG. I see FL-RIG open correctly, and the XML server is on port 12345. But I don't see what combination of settings in FreeData instructs it to use FL-RIG. All I get are error messages that RIGCTLD is not responding. But I don't want (actually cannot use) RIGCTLD. So what is the proper settings to get FreeDATA to use FL-RIG for frequency data and PTT control? RIGCTLD will not work using serial devices.

Is there a plan to implement HAMLIB in addition to RIGCTLD? I know from using other software, such as WSJT-X Improved, the HAMLIB does work correctly. The issue with RIGCTLD is it does not function with Win4Icom. But HAMLIB does work correctly with Win4Icom.

### To Reproduce

Activate FL-RIG, set the XML server to port 12345 (the default as it appears)
Open FREEDATA
Set the rig type to FL-RIG
Set the port to 12345

### Expected behavior

Functional frequency information, and Rcv -> Xmit -> Rcv control.

### Log Files

I can collect the FREEDATA server window data if needed. If I need to turn on extended debugging please let me know.

### Additional context

Icom 7800, FL-RIG is working properly.

Mashintime, 2025-04-21:
There is a FL RIG model in FD (will be within the first few items of the dropdown).  Set everything else to ingnore.  In the custom text box area at the bottom enter -r 'flrig-ip:flrig-port'  (i.e. --r localhost:12345).

DJ2LS, 2025-04-21:
Thanks @Mashintime for providing a workaround.

I'm also thinking about a native flrig implementation. With the given radio control code structure it shouldn't be too hard implementing this. It feels to me, flrig is more used than expected while rigctld/Hamlib causes problems regularly. 
Asking people to use flrig could provide an additional, reliable way of controlling radio

DJ2LS, 2025-04-21:
Thanks for the feedback, @KhunDoug , please also have the look at the PRs, I did some **short** work this evening and we are close to having native flrig support, now. I would appreciate help in testing as soon as I published it.

DJ2LS, 2025-04-21:
@KhunDoug regarding CQ, this is wanted and a feature - when a CQ is received by a station it will repeat in a random time within 10s for avoiding packet colission in case several stations are answering

KhunDoug, 2025-04-22:
Please shoot me an email when you are ready for testing. I know FLRIG is working through external RIGCTLD. I am unable to get internal RIGCTLD to work.

It would be interesting to see if you can get hamlib-xxx.dll working. I think that is the approach used by WSJT-X Improved, and perhaps JS8Call. It works well.

Waiting to hear when I should download and test.

Doug / N3JXB

DJ2LS, 2025-04-22:
@KhunDoug we had lots of problems with Hamlib in the past, specially because of outdated versions . For avoiding incompatibility and blocked serial ports, we moved to rigctld.

DJ2LS, 2025-04-23:
okay, ptt is working as well, power level also. I will publish a test version within next hour. If its working, we can merge it for further testing with more users

KhunDoug, 2025-04-23:
If someone can send me an email with the download link for the updated version with PTT working for flrig, I will test it this week. Email is esunetdude@hotmail.com or n3jxb@hotmail.com. Either will work.

DJ2LS, 2025-04-24:
It's a comment in the FLRIG PR

KhunDoug, 2025-04-24:
The latest patched update resolves this problem. Now flrig and PTT work correctly. I am closing the problem.

closed 2025-04-24 by KhunDoug