DJ2LS/FreeDATA #962: [Bug]: <p> </p> in chat messages

SpudGunMan, 2025-04-22:
### FreeDATA Version

0.17.1 beta

### What operating system are you using?

Ubuntu

### Operating System Version

Mint 5.15 kernel 

### Architecture

x64

### Area of problem

GUI

### Describe the bug

When I get text/QSO from another station. I see the html for paragraphs on the chat text. It’s part of the message in bubble visible, browser is Firefox 

### To Reproduce

Rx any message 

### Expected behavior

The html is removed from the delivered text

SpudGunMan, 2025-04-23:
No just normal hello how are you will pop up with 

<p>Hello how are you</p>

DJ2LS, 2025-04-23:
I'm not sure, why there are html tags inside, its good, the HTML isnt interpreted, so its just text. But until now, its the first time, we have such html tags inside. Is there a chance, you can reproduce it? Maybe the message is some kind of copy  and paste, which injected some html?

DJ2LS, 2025-07-14:
Did this happen again?