DJ2LS/FreeDATA #967: [Bug]: Minor graphical bug when hovering over the "ping" icon

deejdeej, 2025-04-26:
### FreeDATA Version

0.17.1-beta

### What operating system are you using?

Other Linux

### Operating System Version

Debian (Raspberry Pi OS) 12

### Architecture

ARM Raspberry Pi

### Area of problem

GUI

### Describe the bug

Hovering the cursor over the "ping" icon pops up a description message, "Send a ping request to a remote station".  
The pop up message never goes away, even if changing to another screen.

[image]

[image]

### To Reproduce

On the main FD screen, hover the cursor over the "double arrow" ping icon.
The pop up message appears, but never goes away.

### Expected behavior

The pop up message should disappear when you move the cursor away from the ping icon.

deejdeej, 2025-04-26:
Apologies I need to correct the above slightly.  Hovering the cursor over the icon, then moving it away without clicking on it, and the pop up goes away.
However, if you hover over it and the message pops up, then you CLICK on the ping icon to initiate the ping, then the pop up message does not go away.

DJ2LS, 2025-04-27:
using firefox on desktop there is no issue. Question is, if there's another issue which occurs and prevents the popup from getting hidden. Whats the used browser, @deejdeej

deejdeej, 2025-04-27:
The browser is Firefox version 137.0.2 (aarch64).

deejdeej, 2025-04-27:
Also, the Rpi 5 is running X11, and I am connecting to it using VNC (RealVNC server running on the Rpi).

Mashintime, 2025-04-27:
It occurs on chrome/windows too.  I feel like it's a bootstrap issue.

DJ2LS, 2025-04-27:
Oh, wait, sorry, I didn't read your description correctly. I will check tomorrow again

DJ2LS, 2025-04-27:
I pushed a update for bootstrap. Could you test after running a npm run build?

Mashintime, 2025-04-27:
Still occurs, but if you click the pop up once it is frozen, it goes away.  I vaguely remember this being an issue before the first of the major rewrites.

deejdeej, 2025-04-28:
> I pushed a update for bootstrap. Could you test after running a npm run build?

Definitely an improvement.  Now it goes away if I click anywhere on the screen.  The preference would be to go away once the cursor is moved off the icon, without having to click, but this is definitely better than not going away at all like it was before.  :-)

DJ2LS, 2025-05-06:
Can we close this one?

deejdeej, 2025-05-06:
I think the desired behaviour would be to have the pop-up disappear without having to click elsewhere on the screen, if that is possible.  

There are two situations:

1) You hover the cursor over the ping icon in the heard stations list.  The pop up appears.  You move the cursor away from the ping icon.  The pop up disappears.  This seems to work well.

2) You hover the cursor over the ping icon in the heard stations list.  The pop up appears.  You click on the ping icon to ping the remote station.  In this case the pop-up does not disappear, and you have to click elsewhere on the screen to get it to go away.  The pop up should probably go away immediately when you click on the ping icon (before the actual ping starts), or alternatively when the ping finishes.

DJ2LS, 2025-06-01:
I had this issue on iPad as well. But I don't know, how to fix it or create a workaround.