DJ2LS/FreeDATA #977: [Bug]: Some bugs introduced in commit 77b2e9e

OH2JRB, 2025-05-06:
### FreeDATA Version

0.17.3-beta

### What operating system are you using?

Other Linux

### Operating System Version

Debian 12 (Bookworm)

### Architecture

x64

### Area of problem

Server - General

### Describe the bug

```
2025-05-06 14:14:20 [info     ] [RIGCTLD] Connected to rigctld at 127.0.0.1:4532
2025-05-06 14:14:20 [warning  ] [RIGCTLD] Error sending command [\chk_vfo] to rigctld: 'NoneType' object has no attribute 'sendall'
2025-05-06 14:14:22 [info     ] [RIGCTLD] Connected to rigctld at 127.0.0.1:4532

2025-05-06 14:14:44 [info     ] Enqueueing transmit command    command=CQCommand
2025-05-06 14:14:44 [info     ] Running CQCommand
2025-05-06 14:14:44 [debug    ] [MDM] TRANSMIT                 delay=50 mode=signalling
2025-05-06 14:14:44 [debug    ] Event:                         ev={'ptt': True}
2025-05-06 14:14:44 [warning  ] [MDM] [demod_audio] demod loop ended e=ValueError('could not broadcast input array from shape (0,) into shape (8720,)') mode=SIGNALLING
2025-05-06 14:14:44 [warning  ] [MDM] [demod_audio] demod loop ended e=ValueError('could not broadcast input array from shape (0,) into shape (8680,)') mode=SIGNALLING_ACK

2025-05-06 15:33:05 [warning  ] Client connection lost         e=WebSocketDisconnect(1001, '')
2025-05-06 15:33:05 [warning  ] Client connection lost         e=WebSocketDisconnect(1001, '')
2025-05-06 15:33:05 [warning  ] Client connection lost         e=WebSocketDisconnect(1001, '')
Traceback (most recent call last):
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/starlette/routing.py", line 699, in lifespan
    await receive()
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/uvicorn/lifespan/on.py", line 137, in receive
    return await self.receive_queue.get()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/queues.py", line 158, in get
    await getter
asyncio.exceptions.CancelledError

Exception in ASGI application
Traceback (most recent call last):
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/anyio/streams/memory.py", line 111, in receive
    return self.receive_nowait()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/anyio/streams/memory.py", line 106, in receive_nowait
    raise WouldBlock
anyio.WouldBlock

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/uvicorn/protocols/http/httptools_impl.py", line 409, in run_asgi
    result = await app(  # type: ignore[func-returns-value]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/uvicorn/middleware/proxy_headers.py", line 60, in __call__
    return await self.app(scope, receive, send)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/fastapi/applications.py", line 1054, in __call__
    await super().__call__(scope, receive, send)
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/starlette/applications.py", line 112, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/starlette/middleware/errors.py", line 165, in __call__
    await self.app(scope, receive, _send)
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/starlette/middleware/base.py", line 177, in __call__
    async with anyio.create_task_group() as task_group:
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/anyio/_backends/_asyncio.py", line 776, in __aexit__
    raise exc_val
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/starlette/middleware/base.py", line 178, in __call__
    response = await self.dispatch_func(request, call_next)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/freedata/FreeDATA/freedata_server/server.py", line 69, in nocache
    response = await call_next(request)
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/starlette/middleware/base.py", line 148, in call_next
    message = await recv_stream.receive()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/anyio/streams/memory.py", line 119, in receive
    await receive_event.wait()
  File "/usr/lib/python3.11/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError
Exception in ASGI application
Traceback (most recent call last):
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/anyio/streams/memory.py", line 111, in receive
    return self.receive_nowait()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/anyio/streams/memory.py", line 106, in receive_nowait
    raise WouldBlock
anyio.WouldBlock

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/uvicorn/protocols/http/httptools_impl.py", line 409, in run_asgi
    result = await app(  # type: ignore[func-returns-value]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/uvicorn/middleware/proxy_headers.py", line 60, in __call__
    return await self.app(scope, receive, send)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/fastapi/applications.py", line 1054, in __call__
    await super().__call__(scope, receive, send)
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/starlette/applications.py", line 112, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/starlette/middleware/errors.py", line 165, in __call__
    await self.app(scope, receive, _send)
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/starlette/middleware/base.py", line 177, in __call__
    async with anyio.create_task_group() as task_group:
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/anyio/_backends/_asyncio.py", line 776, in __aexit__
    raise exc_val
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/starlette/middleware/base.py", line 178, in __call__
    response = await self.dispatch_func(request, call_next)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/freedata/FreeDATA/freedata_server/server.py", line 69, in nocache
    response = await call_next(request)
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/starlette/middleware/base.py", line 148, in call_next
    message = await recv_stream.receive()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/freedata/FreeDATA-venv/lib/python3.11/site-packages/anyio/streams/memory.py", line 119, in receive
    await receive_event.wait()
  File "/usr/lib/python3.11/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

Terminating application, caught signal 2
Unrecognised PTT type, using NONE

```

### To Reproduce

Got these after git pull origin develop to latest.

### Expected behavior

No errors in log.

### Additional context

This issue created to record this topic for processing in due course.

DJ2LS, 2025-05-06:
Question, did the errors occur on shutdown, by CTRL+C for example?

DJ2LS, 2025-05-06:
One issue ( demod ) occurs with default buffer enabled

DJ2LS, 2025-06-24:
@OH2JRB do we have these issue as well with 0.17.4? Some issues got solved