DJ2LS/FreeDATA #981: [Bug]: FD doesn't automatically retry sending a message

deejdeej, 2025-05-31:
### FreeDATA Version

0.17.3-beta

### What operating system are you using?

Other Linux

### Operating System Version

Debian GNU/Linux 12 (bookworm) on Raspberry Pi 5

### Architecture

ARM Raspberry Pi

### Area of problem

Server - Message System

### Describe the bug

If the first attempt at sending a message fails, FD will not automatically try resending, even after a successful CQ response or ping response.

If you manually click the circular retry symbol next to the message, then FD will attempt a retry.

### To Reproduce

Send a message that fails.
Send a ping to the other station, or send a CW and get a valid response from the other station.

### Expected behavior

FD should automatically try resending the message after a successful ping or CQ response.

deejdeej, 2025-05-31:
Also received a beacon from the other station, SNR 0, and FD did not attempt to resend the failed messages.

Summarizing, receiving a CQ response, ping response, and a beacon does not trigger FD to try to resend a failed message.

deejdeej, 2025-05-31:
In theory this should be easy to duplicate.  Put a dummy load on your station, and send a message to another station that you can normally hit.  Wait until it fails sending the message.  Hook back up to the antenna and then ping the other station.

DJ2LS, 2025-06-06:
@deejdeej any news regarding this? Sorry if I missed a message on Discord