DJ2LS/FreeDATA #1003: [Bug]: Nachrichten in der Warteschlage werden nicht gesendet, man kann sie auch nicht löschen

db5fp, 2025-08-31:
### FreeDATA Version

Last Version

### What operating system are you using?

Windows

### Operating System Version

Win11

### Architecture

x86

### Area of problem

GUI

### Describe the bug

Hi,

I can't clear queued messages and I can't figure out why these messages are not send.
Every thing seems to take long time. Between waht decides to change bandwith of the transmission, we have only manged to geht 500Hz working.

73
Frank
DB5FP

### To Reproduce

Every time, even on old chats with same QSO partner

### Expected behavior

a button to clear even queued messages

### Screenshots

[image]

dk5sm, 2025-09-01:
I can confirm that a queued message cannot be deleted using Windows executable (v0.17.7). The trashcan icon is greyed out.

However, I cannot confirm that the queued messages will not be sent. Here, the behavior is as expected—after a few seconds, the server attempts to send the message.

DJ2LS, 2025-09-01:
Reliable logging on windows isn't working fine at the moment. 
But I guess it's a python error which causes the issue. So if you could just start the server and send me the first 20-30 lines via copy and paste - this should be enough.

db5fp, 2025-09-01:
[neu 2.txt](https://github.com/user-attachments/files/22083376/neu.2.txt)

Ich hoffe das reicht?

db5fp, 2025-09-07:
> @db5fp Sorry for the late replay, I checked your console log and didn't find any reason why the transmission shouldn't start. Maybe there is constant QRM on your channel? If transmissions are not starting, you could maybe post a screenshot of your waterfall screen.

Hi,Is there and chance to set DCD on/off or have an LED indicator for busy channels or even override that?I’ll check for your question next time time an take a screenshot.73Frank DB5FP